nppc/B-uOSD #9: Bill of materials

brucesdad13, 2017-08-09:
Hello! Several of us would like to build one of these and have ordered PCBs from OSH Park. Can you offer a bill of materials of exactly which components we need? There's some confusion amongst us especially on the passives.

nppc, 2017-08-09:
Ok, will add it to the Readme. Do you need some quick answer now?

brucesdad13, 2017-08-09:
Yes a quick answer would be awesome. I'm on Mouser ordering components

nppc, 2017-08-09:
Added BOM.txt to the root folder,
If you have any questions, ask here...

nppc, 2017-08-10:
Added Mouser Part numbers

closed 2017-08-10 by nppc

brucesdad13, 2017-08-11:
Thank you! Will be building next week and maybe have a PR or two if any ideas come up 😄 Otherwise thank you again Pavel this looks like such a cool project. Which AIO are you using?

nppc, 2017-08-11:
Thanks. :)
I tested it with this camera: https://www.banggood.com/Eachine-E010C-E010S-Spare-Part-p-1119217.html
But I think any camera will work. The simplicity of the schematic do not require to cut connectors on the camera. Just solder this B-uOSD to video pin.

brucesdad13, 2017-08-14:
[image]
[image]

Looks ok? 😄 Will try flashing next.

brucesdad13, 2017-08-14:
[image]

It's alive! 😀

nppc, 2017-08-14:
Ah, super! Looks nice. Congratulations! :) 
What camera and vtx are you using?

brucesdad13, 2017-08-14:
This is the Boldclash F-02. It's a little on the heavy side but since comes already split it was the easiest to splice in the OSD. I ordered the exact camera you specified from banggood but it's on the slow boat. 😄 

[image]

brucesdad13, 2017-08-14:
[image]

nppc, 2017-08-14:
Looks nice! How you solder? with solder station or somehow with reflow?

brucesdad13, 2017-08-15:
By hand with a little Zeny hot air and solder 2-in-1 station. 😄 I have two more to build sometime this week. Next I want to look at the code and see... can we fit pilot name?

brucesdad13, 2017-08-15:
Did some testing yesterday 😄 https://www.youtube.com/watch?v=hBjUEI953vA

nppc, 2017-08-15:
Heh... If you will look to the code, then you will see, that we have only 4 bytes (2 instructions) left free in FLASH. :)
So, with current MCU (1KB of FLASH) to fit pilot name is not possible.
But you can do custom bitmap (8*10 pixels) and use it instead of copter bitmap.

And... WOW! I impressed that you did a video!!!
What do you think, can you give me the fragment of your video to post it in the readme here?

brucesdad13, 2017-08-15:
I was wondering what if we removed the serial communications protocol and all the configuration was done compile-time? Yes I will happily give you the video clips.

nppc, 2017-08-15:
OK, please, create new issue, and then I will look how much space we will gain and what we can do with it...

brucesdad13, 2017-08-15:
Created 😄 It might be a bad idea on my part... Thanks Pavel!