nppc/B-uOSD #11: Feature Request: Pilot Name

brucesdad13, 2017-08-15:
Love this project! One idea occurred that perhaps, if it were possible to compile without the serial protocol, we could also have a compile-time option to include pilot name and/or target cursor on-screen.

nppc, 2017-08-16:
Ok. Interesting idea.
Currently we use 1020 bytes of FLASH from 1024. So, no space for new features.
I tried to make conditional compilation without CONFIG feature.
Now we use only 582 bytes of FLASH from 1024. Seems plenty of space! :)

nppc, 2017-08-16:
Alfa version of PILOTNAME feature is implemented. 
`Branch try_name`
I can't test it, because I have no spare pcbs (ordered more boards OshPark)...

brucesdad13, 2017-08-17:
I will test ASAP! I have two PCBs left untouched of my original three. 😄Thanks again, Pavel!

brucesdad13, 2017-08-17:
So far so good. Will build a PCB ASAP. 
[image]

nppc, 2017-08-17:
Thank you.
I got PCBs and assembled one.
Tested... Doesn't work. Need to fix small bug...

nppc, 2017-08-17:
Ok. Bug is fixed! :) Feel free to test working version... [:)[](url)](https://raw.githubusercontent.com/nppc/B-uOSD/try_name/Images/PILOTNAME.png)

nppc, 2017-08-25:
Implemented.
Draft in v1.3 (no official release), in v2.0 (will be released soon).

closed 2017-08-25 by nppc