nppc/B-uOSD #13: Simplify circuit

Dakkaron, 2017-11-26:
This project is amazing! Great work so far!

I think the circuit can be simplified quite a bit though, by leavin out the LMH1980. It is the most expensive part in the design and it's work could be easily done by the ATTiny.

Just replace the LMH1980 with a clamping circuit and use the Analog Comparator included in the ATTiny together with a simple voltage divider to trigger an interrupt each time the clamped video signal dips unter 100mV. This then is the line sync signal. If the line sync signal happens at double speed a few times directly after another, that's the frame sync. On the software side that should not be much of a change.

A similar thing has been done here: https://www.rcgroups.com/forums/showthread.php?1473207-Diy-osd-%28Arduino-and-opensource%29

nppc, 2017-11-26:
Thank you.
Yes, mine first prototype version was planned with discrete components for video sync detection. But it comes in bigger size, than LMH solution. 
If there will be interest, why not to make bigger, but cheaper version?

Dakkaron, 2017-11-26:
Is it really bigger? I just did a quick comparison in components. If you remove the LMH and its supporting components you loose 3 capacitors, 1 resistor and the LMH, but you need to put 1 capacitor, 3 resistors and 2 diodes. As far as I know, all these discrete components are aboutt the same size, so in the end you trade the LMH against two discrete components.

I am by far no expert in making little SMD boards, so I could be completely wrong. But if you use the built-in comparator in the ATTiny, this should work out to be smaller.

nppc, 2017-11-26:
Can you draw something? I have not much time at the moment. If you prepare schematic, I can make PCB. If you like, you can adjust firmware too. :)

Dakkaron, 2017-11-26:
I can try to do something in Kicad (I have never worked with it before, but I'd like to give it a try). If I can't get it to work, I'll send you a schematic.

I have done quite a bit with ATMegas, but only in C. I will have a look what I can do with the assembler code. I sadly don't have much possibility to test this all in hardware at the moment, because I am not near my hardware stash and won't be for a while... I hope, my schematic will make sense...

nppc, 2017-11-27:
Thanks! Schematic and some C code would be good. Then I can find time to make a draft of the new version.

brucesdad13, 2017-11-27:
Excited to see what you come up with! 👍

Dakkaron, 2017-11-28:
I currently don't have access to my hardware and testing stuff, but according to simulation, this here should do the job:

[image]

VIN is the video signal, V1 is the battery voltage (I simulated it with 3.3V and 5V and both seem to work fine, so I guess, there should be no problem with 1s batteries), AIN0 is pin 5 on the ATTiny13A and AIN1 is pin 6.

In the original circuit, R3 and R4 are realized with a potentiometer so the user can find the optimum values for the reference voltage on AIN0, but I think this can be hard-wired. According to the simulations I have done, these values should be ok, but it might be good to check that on a breadboard with a potentiometer first.

nppc, 2017-11-28:
Dakkaron! Thank you! Looks very promising!
Let me find the time to make a PCB... Maybe today... :)

nppc, 2017-11-28:
Full schematic - draft for V2 :)
[image]

Dakkaron, 2017-11-28:
Thanks for integrating that! Looks really good!
I wonder if maybe there should be a resistor between C2 and MISO/PB1. Or do you think the capacitor there is good enough to isolate MISO from VIDEO?

I have not been able to try this in hardware, so maybe it would be good to test this in hardware before integrating this. Especially the values for R7 and R8 should be verified in hardware, but if they are wrong changing the resistors out should be easy to do.

I am really looking forward to this! Thanks for the great work!

nppc, 2017-11-28:
I think AIN1 (PB1) don't need resistor. It is just an input to the MCU.
I'm just curious, do we really need 100 Ohm for R6 and R5? Looks they generating a lot of current?
[image]
Maybe 10K better?

Dakkaron, 2017-11-28:
@Resistor values: you are right, that's a mistake of mine. R6 should be 10K and R5 should be 100K.

@PB1: I thought, that maybe the resistor would be necessary so that the pin would still work as MISO for programming purposes, so that In Circuit Programming works. In your original schematic at https://github.com/nppc/B-uOSD/blob/master/KiCad/Schematic.pdf you secured each of the programming pins with a 120 Ohm resistor. I thought, that was for that purpose, so that the ATTiny / the programmer can drive the pins much stronger than the video signal can.

Dakkaron, 2017-11-28:
I did some more simulations and the values I found to work best with R6 being 10k and R5 being 100k are R7 as 60k and R3 as 40k. I simulated that with a VCC of 3-10V and everything seems to be stable. This way, power consumption should be quite minimal.

It takes a few cycles for the clamping mechanism to work, but the ATTiny-pins allow for a negative current of up to 1mA, so that should not be a problem. That only means that the OSD should wait for a few ms before engaging the interrupt.

nppc, 2017-11-28:
Ok! Thank you for the help! And I have good news :) As you said at the beginning...
[image]
[image]
I tried to put all higher components to one side, and resistors and 2 caps (low profile) on another...
If we look carefully, maybe both caps can be moved to the upper side.

Dakkaron, 2017-11-28:
This looks just cool! Thank you for your work!

nppc, 2017-11-28:
All tall components on one side. Now the pcb will fit between the PCBs of AIO FPV camera! :)
[image]

Dakkaron, 2017-11-28:
That's crazy! Amazing work! Thank you!

Dakkaron, 2017-11-28:
On the software side, what needs to be done is first, replace EXT_INT0 with ANA_COMP (this is all that is needed for horizontal sync). Vertical sync is a bit more complicated, I am not sure, what the best way to implement that is.

V-Sync is basically a H-Sync at double the frequency. A regular H-Sync happens every 63.5us for NTSC and every 64us for PAL. So if a Sync happen quicker than after 50us it must be a V-Sync.

So first you need a global flag bit, I'll call it is_h_sync and a timer that triggers after about 50us.
Whenever a sync happens, check if is_h_sync is set. If it is, treat the following sync as a H-Sync else as a V-Sync. Now unset the flag (set it to false) and reset the timer's counter register, so that it starts from the beginning.
If the timer interrupt triggers, set is_h_sync to true.

This is not perfectly accurate, it will always lag one line behind, but for this application that should not matter. Just don't draw to the first and last lines.

nppc, 2017-11-29:
Ok, sounds a good plan. Will later look to the code, when will have a time...

nppc, 2017-11-29:
Schematic can be finalized I think.
[image]
Maybe I can send the board to OSHPark, to have a prototype for debugging the code...

nppc, 2017-11-29:
Link to OSHPark PCB: [https://oshpark.com/shared_projects/5FUoUMnV](url)

Dakkaron, 2017-11-29:
This looks just cool! Thank you for your work! I'll probably order one of them aswell.

Dakkaron, 2017-12-14:
Just wondering, is there a reason why you are using an ATTiny13a and not e.g. an ATTiny84a? The 84a does not come in the MLF-10 package, but in the MLF-20 package, which has the same size but has pins on all four sides, not only on two sides. Soldering should not be much harder since the pin spacing is almost the sale (0.45mm instead of 0.50mm). Energy consumption is about the same and it is not much more expensve (still below $1 on Mouser). But you gain 6 pins, have 8kb ROM instead of only one and a whopping 512 bytes of RAM instead of only 64 bytes.

With the additional pins you can have the programming pins on dedicated pins which means you don't need R1 and R9 anymore.

And the additional RAM and ROM should allow for more advanced features like reading telemetry from a FC or fonts in multiple sizes.

According to the information I found, the architecture of both chips should be the same so you should be able to use the same software you use now, only needing to change pin assignments.

nppc, 2017-12-14:
Hi Dakkaron! Yes, there was a reason. 
Attiny13a package is 2.9x2.9mm is size, where Attiny85 has a size of 4x4mm. As the project was initially planned as a simple micro OSD, the size of the mcu was more important than a size of a flash.

BTW I had a plan to make bigger pcb with Attiny85 and 20mhz oscillator, but currently I have no time to work on it.

nppc, 2017-12-14:
BTW useful pin count is the same for attiny13 and attiny85
[image]
[image]

MicroMontis, 2017-12-14:
So we can just swap Attiny13 with Attiny85? I would stick with Attiny85... just in case we will need more program space :)

nppc, 2017-12-14:
Yes, but need bigger pcb, because footprint of attiny85 is 25% bigger than attiny13.

Dakkaron, 2017-12-14:
I am sorry, I was talking about the ATTiny84a, not the 85. The 20M2 package is 3x3x0.85mm, which is about the same as the ATTiny13a.
[image]

Also it adds more usable pins:
[image]

nppc, 2017-12-14:
Ah, ok! I'm overlooked this. I thought about attiny85, but you wrote attiny84A :)
Yes, why not to replace attiny13a with attiny84a... I think this is very good option!

But we need carefully to think, what we do with this. 
For example, 

- I would like to run it with external oscillator, and if possible at 20mhz (gives better stability). Only questionable, how it will run on 1S at 20mhz?
- Then maybe we can implement some sort of serial communication to change settings, and get data from FC. (maybe easiest way to parse MSP?).
- With this chip it is really possible to make small FC with integrated uOSD...

But I have no time for everything :(

Dakkaron, 2017-12-14:
* I would not recommend running 1S @ 20mhz. 1S can easily go down to voltages as low as 3V with full throttle. The maximum safe operating frequency for 3V is around 11mhz. According to the datasheet 20mhz requires at least 4.5V. So for that we'd need a voltage booster. The problem is, voltage boosters for 5V are quite hard to come by and usually comparatively bulky (at least from my experience, I'd like for someone to prove me wrong ;) ).

* Parsing MSP should be rather straight-forward if there is more space on the device. I found an Arduino-MSP-parser here: https://github.com/fdivitto/MSP . Looks rather short and simple, so it should not be too hard to implement something like that into B-uOSD.

I'd like to contribute code once my hardware arrives. But I can't make KiCad schematics, since I have no experience in that.

nppc, 2017-12-14:
Yes, 20mhz kinda Scarry. But 16 I'd like to try. Atmega runs 16 on 3.3 with no problem.

I thing C1 should be there. Bigger is better, but if you have something smaller, it will work too. 
It is for stabilizing the voltage on MCU. We have analog circuity there, so decoupling is essential.

nppc, 2017-12-14:
To implement serial protocol is quite tricky. We have not much power for that. I think we can't use any interrupts while drawing OSD, otherwise symbols will be jittery.
So, we can parse serial only within very limited timeframe. Don't yet know is this possible at all or not?

Of course, we can use serial only for configuring, so need to evaluate it.

Dakkaron, 2017-12-14:
The ATTiny84a has one USI interface. That's a generalised form of a hardware-accelerated serial interface. This reduces the overhead of serial communication by quite a bit as you only have to poll once 8 bit have been sent.
PAL has fewer lines per second than NTSC, so if we want to read one byte per line max, then we can read 15625 bytes per second. That means we could read up to a baud rate of 125000.
MSP seems to be running with a baud rate of 115200, so we should be just about fine.

nppc, 2017-12-14:
The implementation of UART with USI interface needs quite an amount of CPU time (see AVR307 app note). First, we need pin change interrupt to detect incomming data. Then start a timer to get timing for USI, and update timer every received bit. So, to do it while drawing the text is not possible I think.
More over, if I remember right, then MSP protocol requires first to request data, so we need to send data first. Then we don't know exactly, when data will start to come.

So, maybe it will possible to send/receive data, while we not drawing anything.

Please correct me of I'm wrong somewhere...

Dakkaron, 2017-12-15:
From what I understand of AVR307 the biggest problem with UART over USI is that UART packages can bei 7 to 11 bit long, while USI only has the infrastructure for 8 bit packages, which then requires the CPU to do a lot of work to work around this limitation.

I have read somewhere that MSP can also be sent over I2C/TWI, which is a lot easier to implement on USI because it is 8-bit-aligned, same as MSP. I2C over USI is detailed in AVR312, but there they use an interrupt to read the packages. As an interrupt is not feasable in this project, there is a way to do this with polling. If the interrupt is disabled, the USIOIF bit in the USISR register indicates that 8 bits have been read and the result has been copied from USIDR to USIBR. Using the USI Buffer Register instead of the USI Data Register to read data means that there it is not necessary to read the data before the next bit arrives, but before the next byte arrives. At a clock speed of 115200 this happens at a maximum of 11400 times per second. We have 15625 lines per second in PAL and a slight bit more in NTSC, so that should work out.

As for writing, I imagine we are in control of when we write, so we could just do that during field sync since we don't have to do anything else there.

If that does not work out, we could just dedicate one field per second for reading and writing. Since most FPV receivers have do do deinterlacing anyways that wouldn't even cause the characters to fully disappear during that frame, but only every second line.

In any event I think the ATTiny84a would be a good choice since it does not cost much more and would add a lot of flexibility, no mattter what will be done with that flexibility afterwards.

MicroMontis, 2017-12-15:
I think we should first get a prototype running, just to justify we can do OSD without video sync chip.

Dakkaron, 2017-12-15:
I'm just thinking, the ATTiny13a is currently alsmost at its ROM limit with the current state of things. Tha ATTiny84a would make it easier to integrate the additional code necessary for integrating video sync.

nppc, 2017-12-15:
Yes, we can test the "video sync less" design with attiny13a because PCB is already on its way from OSHPark, and then we can move to attiny84a as a next step.

@Dakkaron If you like, feel free to start new issue for attiny84a and lets look closer to it after attiny13a version is working. What do you think?

Dakkaron, 2017-12-15:
Sounds good. I just feared that the "video sync less" design might not fit onto the attiny13a's ROM. But if you think this works this sounds like a good plan. Migrating from ATTiny13a to ATTiny84a should not be too much work after that (at least in theory).

nppc, 2017-12-15:
Space currently should not be a problem. with LMH I was at 75% of Flash, so, 25% is quite a big amount in assembler :)

MicroMontis, 2017-12-15:
OMG. I forgot, that it is written in ASM. Respect.

Dakkaron, 2017-12-15:
Oh, that's good! I thought I read somewhere in the commits that it was at +90%.

vidzo, 2018-08-28:
I have recently found this great little project and have ordered a few V2 boards. I now that nppc  is a bit busy with other projects but was hoping there had been some progress with the code? @Dakkaron I'm more than happy to help with testing once I receive and make the boards. I know very little code though...

Dakkaron, 2018-08-30:
@vidzo Thanks for the interest! I should really build my boards as well. I have them lying around since a few months. From what I understand, there is really not much left to do. The code should be basically working. The main problem is figuring out which resistor will work for the analog comparator.

vidzo, 2018-08-30:
It's a great little project, how could I could not be interested?! I'm more than happy to offer some testing time with various resistor values. I'm going to order some resistor kits in the next few days along with the MCU. Hopefully Oshpark boards come in the next few weeks too

erneut, 2018-09-03:
@Dakkaron Hello. Can you please tell me the way you've tested circuit you attached [here](https://github.com/nppc/B-uOSD/issues/13#issuecomment-347494728)? I'm trying to understand the way this circuit works but I have no idea how to simulate input signal :). It looks like you used CircuitLab. Did you use "Behavioral voltage source"? If you did what kind of expression did you use? Thank you

vidzo, 2018-10-28:
Hey nppc, have you managed to write up a BOM yet?

nppc, 2018-10-29:
Sorry, I forgot about it... Thank you for the reminder... Will do it as
soon as I can... :)
Pavel

Dakkaron, 2018-10-29:
@erneut Sorry for the late answer. I used CircuitLab and I tested it only using sine wave inputs. I didn't come up with the circuit myself, I just copied and modified it from here: https://www.rcgroups.com/forums/showthread.php?1473207-Diy-osd-%28Arduino-and-opensource%29

The main difference between the original circuit and this one, is that the poti was changed to a fixed resistor, since a potentiometer would make the module too large. But maybe a tiny trimmer potentiometer lieing flat on the back of the module might work. They are still way cheaper than an LMH1980. But it would be better if it could be done without one.

nppc, 2018-10-29:
BTW: For testing purposes you can omit R8 and use external potentiometer connected to the MOSI pin of the board and to the ground. After finding working value you can solder it in place of R8...
[image]

nppc, 2018-10-29:
BOM is updated for V2: https://github.com/nppc/B-uOSD/blob/testANACOMPsync/BOM.md

vidzo, 2018-10-29:
Thanks Pavel, will try and get this built next month for testing

vidzo, 2019-02-10:
Hi Pavel, is there a image of the PCB with a parts placement identifier that corresponds with the BOM? Attempting to place the resistors on the bottom of PCB but there is an additional 220(ohm or K?) resistor that isn't listed in the BOM. And not sure where the 62k resistor is meant to go? Instead of the 80k. I'm not great at reading schematics. Thanks

nppc, 2019-02-12:
Hi @vidzo!
So, 220 means 220 ohm. You should really play with R7 and R8 to get good sync results. So, R7 is 62K and R8 need to find a good value.
[image]

[image]

vidzo, 2019-02-12:
Hi @nppc,

Thanks for the reply. I was going quite another content this morning saying that I managed to work it out myself. I went through the schematic and datasheet with a multimeter I worked backwards. I learnt something in the process. But thank you.

I'm actually now at my next issue. I've connected it all to my usbasp but do I also need to power the board externally from a battery/source? It's currently not connecting  when using AVRDUDE. Also, I'm not really understanding how you compile e.g pilot name etc?

nppc, 2019-02-12:
Yes, the board should be powered. I think I powered it with supply power from programmer. And also re-check connections as this board is really tiny.
And check the error message you get. As it can be issue with the driver too.

Compiling your name is a bit tricky as mcu has not much flash memory. 
You need to define letters (lines 50-54 in main.asm) what you will need. If letter appears twice, define it only ones. 
Next you need to define order of characters in your name (line 287 in main.asm).
And in line 253 of the main.asm you should put length of your name (not longer than 10 chars).

If you want to enable/disable information on the screen or change the position of the element (like enable crosshair or disable timer, then you need to adjust lines 247-280. Comment out not needed block or uncomment it if you need it. And adjust column and line for the position of every block.
And at the end make sure that in line 247 you have correct quantity of data blocks (in current firmware there is 4 blocks: pilot name, timer, voltage and minimal voltage).

Hope this helps.

vidzo, 2019-02-12:
Thanks. I'll have to look into it all. I've connected all the pads and am powering it through the VCC of the programmer. Is this correct?

My confusion is to what to change and compile but I'm assuming that main.asm is edited through Notepad or a similar program? Thanks, will study it all and have another go.

nppc, 2019-02-12:
yes, VCC should be fine.
you can edit main.asm in notepad and then compile it as it described in readme.

vidzo, 2019-02-15:
Thanks nppc. I managed to connect to the chip (I forgot to add a jumper for power on the usbasp), compiled and flashed. I compiled a V2 hex (which has less code lines). However, after attaching to the AIO (LST S2) I am not getting anything on the display. I am using the top pin of the three on the VTX side of the AIO. This is meant to be signal. Is this correct? Also powering the Attiny13a through the ground and VBat pins on the VTX side. Are these the correct connections?

Also, when commenting in the main.asm file, when enabling the byte selection are supposed to comment in ;print name to;print name or ;print timer to print timer etc? Slightly confused by that

nppc, 2019-02-15:
You should compile it from **testANACOMPsync** branch (https://github.com/nppc/B-uOSD/tree/testANACOMPsync/VmOSD/VmOSDV2).

V is the video signal on uOSD board. It is top on both sides. I'm not sure where is the video on your AIO. On mine AIO it is the top one of three pin connector.

You can power the board with anything from 3 to 5 V on the +- pins. 5V from VTX is good. But I'm not sure what pin is Vbat on your VTX.

for commenting (disabling) lines you use ;.
[image]

[5 more replies]